fluidd-core/fluidd #5: [Proposal] UI Change

obelisk79, 2020-10-18:
Hi, talking with you on Discord I mentioned providing a mock-up of some UI layout changes.

The thought process here is to keep the static sized dashboard items completely static so that other panels of variable size don't move them around. Also note the location of the icons for the macros/jobs/commands/console panel keeps it so you can click through them quickly.
[image]

cadriel, 2020-10-18:
Some context to this after some discussion;

It feels like some would prefer the temp status and some controls further up in the panel hierarchy.

Rather than radically changing the current layout, we could instead look into other options, like;
- Allowing panels to be expanded / contracted - i.e., the tabbed cards could be contracted by a user, bringing the temperature card further up the fold when needed.
- Moving the tabs to the top of a card, instead of the bottom.
- Looking into making the tabs themselves a bit smaller.
- Ensuring that each tabbed card is the same height, instead of them pushing content up and down as you change tabs.

So - i'll look into these options and move forward with one or many.

Note that having the printer status panel at the top left is something I'd want to keep in place, along with the main tabbed card at the top right if possible - due to these being priority for a user before and during a print. I.e., I feel like not having to scroll for macros and / or the console and jobs are important to retain.

obelisk79, 2020-10-18:
Good points

obelisk79, 2020-10-25:
So, bringing over from discord discussion.

I also had an idea which may be beneficial for mobile users with limited screen real estate. To add bed/nozzle temps to the status panel. This along perhaps with other print metric information could be controlled via setting toggles to allow users to control which information they have a quick-view of.

obelisk79, 2020-10-25:
Another feature/tweak

Something Mainsail currently does is remove underscores from macro text in the UI buttons. This should be an easy modification that just adds a bit of polish to the presented interface.

cadriel, 2020-11-03:
Im going to close this issue for now - as most topics have been covered.

closed 2020-11-03 by cadriel