fluidd-core/fluidd #6: Setting chamber temperature

khumarahn, 2020-10-18:
Hi! I installed fluidd to look at it. The first thing I noticed is an error on trying to set chamber temperature:
```
{'message': "Malformed command 'SET_HEATER_TEMPERATURE HEATER=heater_generic chamber_heater TARGET=50'", 'error': 'WebRequestError'} 
```
This is similar to the issue in mainsail: https://github.com/meteyou/mainsail/issues/87

Also, `heater_generic chamber_heater` could be shortened to something readable.

[image]

cadriel, 2020-10-19:
Yep, I'll take a look.

cadriel, 2020-10-19:
@khumarahn Attached is a copy of fluidd with updates to handle generic heaters.

Are you able to;
1. Ensure you can now set temperatures properly and;
2. Ensure the graph updates properly for both the target and current temp.

The easiest way to test this is to delete the files in your existing fluidd folder, and unzip this in place of them. Depending how you installed fluidd, it is most likely found here; 
```
cd ~/fluidd
```

Thanks!
[fluidd_test.zip](https://github.com/cadriel/fluidd/files/5403410/fluidd_test.zip)

khumarahn, 2020-10-19:
Hi. I did the checks you asked, it looks like it works correctly! Thanks a lot!

cadriel, 2020-10-20:
Yup, I'll close the bug - the fix will be in the next release.

closed 2020-10-20 by cadriel