fluidd-core/fluidd #8: Feature Request: Filament Usage Tracker with Library

Annex-Engineering, 2020-10-19:
First of all, thanks for developing this, its nice to have some options available on the playing field.

Thoughts on adding a filament usage tracker. Ive been using this plugin on octoprint for a few years, its quite useful when you have 20-30 spools of varying colors and volumes remaining.

https://plugins.octoprint.org/plugins/filamentmanager/

cadriel, 2020-10-19:
Currently, fluidd's backend is moonraker - and a json file for settings.

In order to track things like this - i'd need some way of tracking data more reliably than a json file, and I don't think it's necessarily reasonable to expect moonraker to handle these tasks.

So - we might need to put this enhancement into a holding pattern until I can better understand how to approach this - as i'm sure other requests are going to come through with similar needs.

wlhlm, 2021-03-05:
Moonraker recently received support for persistent database storage, which is a major step towards implementing filament usage tracking.
https://github.com/Arksine/moonraker/commit/54b1bd8b0404f842828f3a0df6d7662c4c9a9648

cadriel, 2021-03-05:
Yes. Fluidd will be moving to the new database storage in the next few releases. Secondarily, filament tracking is on the roadmap as a moonraker plugin - so all clients can make use of it.

mateuszbrzezinski, 2021-03-14:
That might be helpfull for this task: https://github.com/Arksine/moonraker/pull/109

mateuszbrzezinski, 2021-03-17:
Actually started to working on this plugin, first have some initial implementation for moonraker plugin, but I'm rather backend programmer, so looking for someone who could do fronted part of this in Fluidd

matmen, 2023-08-16:
I think the addition of Spoolman (https://github.com/fluidd-core/fluidd/pull/1119) solves this, so I'll go ahead and close this ;)

closed 2023-08-16 by matmen