fluidd-core/fluidd #10: Question: Is there a way to disable some macros while printing

dushyantahuja, 2020-10-22:
Just wanted to check - I have a few macros, that would create issues with prints if they are run while printing. Is there a way to tag macros so that they are disabled while printing.

cadriel, 2020-10-22:
Not currently. I've had people suggest both scenarios whereby some want them disabled, versus not during a print.

I'll have a think about this. There are other enhancements to macros regarding organization and categorization - and that might be a good place to implement something like this against too.

dushyantahuja, 2020-10-23:
Sounds good. Thanks for your great work. I had never thought that I could uninstall octoprint.

cadriel, 2020-10-23:
No problem. :)

cadriel, 2020-11-03:
Im going to close this for now, but improvements to macros and their display is on the list of improvements.. so watch out for a future release.

closed 2020-11-03 by cadriel