fluidd-core/fluidd #11: Infinityh NaNm NaNs in print timer

LazaroFilm, 2020-10-22:
`0h 2m 22s / Infinityh NaNm NaNs` shows in the print timer section. 
My Pi is on the correct time zone (checked, and did a full reboot) 
This happens with any files made from SuperSlicer (have not tried other slicers) 
Mainsail is able to read the times properly. 

[image]

cadriel, 2020-10-22:
@LazaroFilm What estimations are you using? I.e., on the settings page - you have many options for choosing estimations, which one is set?

Also - if it is set to slicer, have you configured super slicer to send estimations?

Does the issue go away if you choose something other than slicer from the settings?

LazaroFilm, 2020-10-22:
I'm using `Klipper estimations (accuracy over time)`
I have `Supports remaining times` selected in SuperSlicer
So I rotated through the settings and the NaN error disappeared from all settings (even the original setting) after that.
I'm currently in a print. I'll try to start the next print with a different Time setting to see if it does the same thing.

cadriel, 2020-10-30:
I've added some extra checks here, can you please confirm and close with v0.0.9?

LazaroFilm, 2020-10-31:
Just updated tonight. Will test tomorrow.

LazaroFilm, 2020-11-02:
Still happening on v0.10 when Klipper is the time source. Other time sources are working fine.

AlexGirardDev, 2020-11-02:
I have also been running into this issue since I want to say 0.06,

I am user SuperSlic3r as my slicer and have updated moonraker and fluidd to the latest versions

cadriel, 2020-11-03:
Is it for any size print? Or super long ones?

AlexGirardDev, 2020-11-03:
All sized prints, ranging from like 6 minutes to 40 hours.
I have attached a sample gcode from my slicer that repro's the issue on my end.

If you are unable to repro this yourself I can deploy some debug builds if you have them or I can also just try and reformatting my PI with a fresh install to since this problems seems to be a config issue at this point as it only effects a small nuber of people.

[left-plate.zip](https://github.com/cadriel/fluidd/files/5479282/left-plate.zip)

cadriel, 2020-11-04:
I've tried with your gcode, and am unable to reproduce.

I'll have a deeper look into this over the next day or so and see if i can't spot anything obvious.

cadriel, 2020-11-04:
Hrm, I don't suppose you guys are using Internet Explorer are you? If so - what version?

cadriel, 2020-11-04:
Ok, I think I may have found a culprit.. 

Can you give this test build a shot and let me know how you get on?
[fluidd_v0.1.0_test.zip](https://github.com/cadriel/fluidd/files/5485448/fluidd_v0.1.0_test.zip)

AlexGirardDev, 2020-11-04:
Yep this seems to have fixed the issue!

FWIW I am not using internet explorer lol.

Thanks for your awesome work @cadriel and tracking down this bug.

cadriel, 2020-11-04:
Fantastic news! I'll merge the change, you can expect it officially in the next release. :)

closed 2020-11-04 by cadriel