fluidd-core/fluidd #15: Feature request: Setting to invert Z control

cjackson234, 2020-10-25:
Issue: My railcore (and quite a few other designs) have beds that drop rather than a gantry or X axis that is raised. So for me, up and down are backwards on the toolhead control.

Suggestion: I would like to see a setting that allows me to invert this control and save it to the settings json.

cadriel, 2020-10-25:
Yep, sounds super valid. Will get it done!

cadriel, 2020-10-27:
Can you clarify exactly how you'd like to see this?

I.e., do you want the icons to stay the same - but the gcode sent to be reversed?

Also - do you want the same application for X axes?

cjackson234, 2020-10-29:
Yes to both. I want the icons to stay the same on the UI and the command behind them to be inverted.

cadriel, 2020-11-08:
This is implemented for the next release. You'll be able to invert each of the configured axis.

closed 2020-11-08 by cadriel