fluidd-core/fluidd #21: All temps in graph

LeandroMarceddu, 2020-10-30:
As title says. Is there a way to display all the temperatures it senses in the graph below it? As seen in screenshot I have a bedcorner sensor, which isn't displaying in the graphs.
[image]

cadriel, 2020-10-31:
Possibly.. can you paste the section of your config where you define this sensor?

cadriel, 2020-10-31:
Also, does the current temp in your screenshot have a min / max value tooltip?

LeandroMarceddu, 2020-10-31:
> Also, does the current temp in your screenshot have a min / max value tooltip?

It does. Which rather confused me at first before I knew what it was about. 

> Possibly.. can you paste the section of your config where you define this sensor?

[temperature_sensor bedcorner]
sensor_type: NTC 100K beta 3950
sensor_pin: P0.24

cadriel, 2020-11-02:
This is fixed, will be available next release. :)

closed 2020-11-02 by cadriel