fluidd-core/fluidd #33: Allow rotation of camera image

cadriel, 2020-11-29:
Possibly use black bars and an aspect constrain of some sort.

aepkolol, 2021-07-02:
I tried to use '-rot 90' in /boot/fluiddpi.txt and it doesn't make a difference

tjengbudi, 2021-08-20:
is there any eta for this?

sdc53, 2021-09-04:
FYI this is editable in webcam.txt from the fluidd gui:
camera_raspi_options="-x 1920 -y 1080 -fps 30 -rot 90"

tjengbudi, 2021-09-04:
@sdc53 looks like it is not from gui right? if i am not wrong you state webcam.txt, this is mjpegstreamer file file configuration right? it is used driver/hardware capability to used it right?
the first problem is i used logitech c270 instead raspi cam. the driver said it cannot rotate. that's why i asked to rotated within web code (html5 example by mjpegstreamer can do it) so the second problem is, looks like fluidd use vue, i am not understand at all. the third problem if i rotate the camera, it will crash the bed. another position is not an option cause i used enclosure and i might dumb at design also

closed 2022-02-20 by matmen