fluidd-core/fluidd #34: Feature Request: Adjust mobile page and etc.

system3rd, 2020-11-30:
first of all, thank you for your effort and fluidd is the best web UI for 3d printer.
I have a few Feature Request since I used it a month.

1. Adjust mobile page button (emergency stop and menu)
- due to my finger is not accurate to touch small button, emergency stop and menu button is too close to make mistake.
 it would be appreciated to make some distance between them.

2. Printer's name on mobile page top bar
- I have 3 printers and sometimes it is confused which printer I controlled.
 Differently from PC web page, mobile page's top bar is not shown printer's name.

3. Probe_calibrate adjust function.
- Once I put probe_calibrate gcode, I have to type repeatly testz z=+- a lot for calibration.
and it is not easy to type in mobile UI.
it would be appreciated to make testz button like babystep button.

Thanks !!!

cadriel, 2020-12-03:
The latest release has #2 implemented.

#1 I think will be adjusted with upcoming updates to the nav.
#3 is on the list! :)

cadriel, 2020-12-09:
Im going to close this because most issues raised are either coming, or are already done.

Thank you!

closed 2020-12-09 by cadriel