fluidd-core/fluidd #36: Is there anyway of adding a power button?

krpepe, 2020-12-03:
Hello,

Thank you for the wonderful interface for klipper! I absolutely love it! One question, is there any way to add a power button that would work with  https://github.com/Raabi91/powermanager. Would like to have the ability to turn on the printer from the interface.

Thank you in advance,

Kevin

cadriel, 2020-12-03:
klipper has a power plugin, which fluidd works with already.

closed 2020-12-03 by cadriel

krpepe, 2020-12-04:
Any possibility you could point me to what you are talking about? Apparently I am blind or stupid because I haven't been able to find a power button on the user interface.

Kevin

cadriel, 2020-12-04:
https://github.com/Arksine/moonraker/blob/master/docs/configuration.md

That explains how to setup the power plugin in moonraker. Once you have that configured, then the buttons will show up inside Fluidd.

krpepe, 2020-12-04:
Thank you very much! I truly appreciate the help! Have a great weekend.

Kevin

lalek54, 2020-12-21:
Hi. 
i tried to configuré à wifi plug. 
But power.py can’t init due to lack of gpiod module in moonraker-env on fluiddpi image. 
Due to that moonraker server is shutting down. 
installing gpiod in moonraker env solve the problèm. 

And also can you tell me where should be the « power printer » button in fluidd. I don’t see it.

lalek54, 2020-12-21:
Problème solved. Thank for your incredible work.