fluidd-core/fluidd #38: Save Settings / ViewEdit Configs With Reverse Proxy

H2OldManZ, 2020-12-07:
Hey!  First off, great work on this!  

My issues stem from using my Proliant server to proxy traffic to my Pi3  where Fluidd lives on the printer.
This works as intended for Mainsail, but with Fluidd I am unable to saver any setting to the UI.  I am also unable to upload gcode or edit/view any system files.  When I connect directly from my LAN it works as intended.

Thanks for looking!

cadriel, 2020-12-09:
I'm not sure I understand the problem.. 

Fluidd saves / loads files the same way mainsail does, i.e., via a call from Moonraker. There shouldn't be any difference there.

Fluidd DOES allow you to connect to your printer instance directly to the moonraker socket, instead of via the nginx proxy tho - so depending on how you're connecting, that may make a difference.

However - in both cases - this should all still work with a correctly configured proxy.

cadriel, 2020-12-09:
You might want to open the hamburger menu (top right) and investigate how you're connected to your moonraker instance (i.e., via `host:80` or `host:7125`) and see if that makes a difference.
[image]

H2OldManZ, 2020-12-09:
Thanks for the response! 
I am having issues adding a new printer in that menu.  
I will do a deep dive into this myself.  
With the info you gave, I suspect it is an issue with the way Kiauh sets up the Default Nginx Proxy settings.
Thanks for the info!!

closed 2020-12-09 by H2OldManZ