fluidd-core/fluidd #49: Sheet management

cadriel, 2020-12-23:
Create a sheet management feature, whereby users can select a sheet prior to printing that sets a specific z-offset.

- Should probably be something users can toggle on an off via settings
- Should be able to add / remove / duplicate sheets
- On print start, should require the user select the correct sheet to avoid mishaps

garethky, 2021-02-22:
+1

The big win here is saving the Baby Stepping value as its adjusted so you don't have to constantly edit a config file. I want to use Klipper on my Toolchanger and I don't want to have to manually enter values for 4 tools x 3 sheets. It needs to be automatic like my Prusa MK3.

garethky, 2023-01-07:
I've written a generalized build sheet manager and "adjust live-z" macro:
https://klipper.discourse.group/t/build-sheet-manager-adjust-live-z/4013
source here: https://github.com/garethky/klipper-voron2.4-config/blob/mainline/klipper_config/build-sheets.cfg

The only "gotcha" is the way klipper handles the offset. Its not possible to say what part of the offset is from the build sheet tool and what part might be from other sources. As such you are required to reset the offset (either via a calibration tool or to a fixed value) at the start of every print. Because people operate printers with very different homing and calibration sensors and routines I went for a solution that would support everyone.

its working great on my Voron 2.4 and I have had a much improved experience with using different build sheets.