fluidd-core/fluidd #52: Update XYZ coords to be inputs, in order to send the toolhead to a specific location.

cadriel, 2020-12-24:


Taudris, 2021-01-05:
Bunch of random thoughts about this feature:
* This feature should allow clicking a separate button to move all axes at once.
* It should also include E for completeness' sake.
* The XYZE inputs should accept absolute or relative positions. Users make it relative by prefixing + or -.
* The feedrate should be adjustable.
* The inputted value should remain so if the movement needs to be repeated, users can press enter/click a button again to repeat it. (This means the current coordinates can't simply be shown in the input; there needs to be separate input and display.)
* The inputs should be disabled during a print.
* It would be cool to have a view of the bed plate where you can click/drag to move the print head around in addition to directly entering coordinates. Visualize the gantry as a pair of parallel lines with a box for the hotend and a spot for the nozzle (with configurable sizes/offsets of course), and ideally see the job's current layer (could be added later).
* Somewhat related feature: add a history of movements in gcode format which can be copied to the clipboard or saved. This could be used to make purge or test patterns.

cadriel, 2021-05-07:
Closing this for now. The latest release reflects this request.

If you have further comments, or would like to suggest more - please feel free to comment here or join the Discord!

https://discord.gg/GZ3D5tqfcF

closed 2021-05-07 by cadriel