fluidd-core/fluidd #53: Delta Printer Friendly UI

gjudianto, 2020-12-24:
Is it possible to change the axis home buttons that are friendlier for a delta printer? Home A, Home, B, Home C, Home All.

[image]

cadriel, 2020-12-27:
Does it make sense to have the individual home buttons at all?

I.e., you can't home each axis on a delta with Klipper right? Would it not make sense to just have a single home button for the 3 rods., and then a single home button for Z?

Can you confirm that homing Z independently is still a thing on a delta configuration?

If so, perhaps it makes sense to just retain the two home buttons on the main panel and remove the individual home icons in the title for delta's.

Hywelmartin, 2021-01-04:
no independent homing on deltas.. One button is enough... in the middle of the "cross"

cadriel, 2021-01-16:
this will be in the next release! let me know if there are issues. Thanks!

closed 2021-01-16 by cadriel