fluidd-core/fluidd #59: Overflow in tools card

davidsteinberger, 2020-12-31:
I have quite some features of klipper enabled. That overlows the tool section.
[image]

Could we add some overflow here (can send a PR for that)?

cadriel, 2021-01-02:
Yep. I was hoping people wouldn't need all of these things at the same time. I knew it'd become a problem 👍🏼 

I'll look into this, but likely I'll condense these buttons back into the hamburger I think.

cadriel, 2021-01-03:
Ok - I've put a temporary fix in whereby the menu condenses to the hamburger at a higher breakpoint now. This will come in with the next release.

In a future update I'll likely move the home buttons out of here to alleviate this further.

Please re-open this bug if this doesn't fix the issue for you!

Thanks!

closed 2021-01-03 by cadriel

davidsteinberger, 2021-01-06:
@cadriel Awesome, Thank You!