fluidd-core/fluidd #60: FR: Temperature Graph Scale Adjustment

J4CKT, 2020-12-31:
Thank you for the awesome interface.

Would it be possible to add scale adjustment of the temperature graph to 20 / 30 / 60 / 120 minutes etc so you know how long you have been preheating.

Would be very helpful especially for those with large printers like my 350 v2 that needs over an hour heatup for a consistent first layer.

Thanks!

cadriel, 2021-01-10:
- X axis is directly effected by width. You lose resolution as the width of the graph is reduced, so adding further time to the scale here would adversely effect the reason the temp graph exists.
- I feel like relying on the temp graph to determine how long you've been preheating is the wrong approach.

Closing as wont fix for now because of the above reasons.

closed 2021-01-10 by cadriel

cadriel, 2021-02-16:
For those involved, this is coming in for the next update.

You'll be able to define the length moonraker stores temperature values - and Fluidd will respect that setting in its thermals graph.