fluidd-core/fluidd #65: FR: Add Save baby step to printer.cfg for easy z end stop calibration Prusa style

GitjoPowershell, 2021-01-02:
after doing a z endstop calibrate we have a rought position of the z that need to be adjusted with baby stepping. I then edit the endstop position in printer.cfg. Does it sound possible to implement the following in fluidd : allow a save of baby stepping that effectively affect z endstop position in printer.cfg 

Already discuss here : 
https://discord.com/channels/460117602945990666/776483964587474974/780405538831335464

But I would like to add it here to avoid forgetting this one

shinuza, 2021-01-04:
The feature is nice, but this would most likely require that the `printer.cfg` is edited in place, e.g downloaded, parsed, edited, and written.
The other way would be through `Z_POSITION_ENDSTOP` but afaik that implies a `SAVE_CONFIG` which also implies a `FIRMWARE_RESTART`. And, I don't think you can run that on a running printer.

So, I guess we should probably ask upstream to have ways to that via the API in Moonraker. What do you think @cadriel  ?

cadriel, 2021-01-04:
My plan is to expand the printer configuration page to include setup controls like this. 

I think this would work similarly to the bed mesh config.

There's a property that moonraker exposes that tells us if klipper has unsaved changes. I.e., if you calibrate a mesh, or adjust your z offset - that I haven't taken advantage of quite yet, but is a good candidate for this.

I.e., we have a card that allows the user to jump into an endstop calibrate mode. It has buttons that allow the user to set testz up and down - and a save option that then restarts the printer with a warning about losing unsaved changes in your config.

Basically, UI that removes the need for users to type those commands themselves in the console - since we know they'd have to do this anyway.

An extension of this is then the issue related to sheet configuration. I.e., a way for the user to preselect a sheet with an offset for quick changes. I think the first endstop calibration would define the base sheet, which others inherit off'f. 

Suspect these are all related;
https://github.com/cadriel/fluidd/issues/57
https://github.com/cadriel/fluidd/issues/49

GitjoPowershell, 2021-01-04:
Cadriel, this is indeed awesome can't imagine that you plan to expand that much the proposal,

The offset configuration card is a good idea, however it does not allow you to save babysteps that you change during a print, which was the original Idea of this feature request.

The plate swap feature (like swap bed meshes) is awesome :) 

do you think feasible to implement a babytsteps save into printer.cfg ? alongside the offset configuration card it could be huge :)

Best regards,

cadriel, 2021-01-04:
Saving the adjustment during a print to your printer isn't possible, nor recommended. I think maybe taking note of the value during a print in order to apply once the print is complete is more feasible - which is really what the feature I was talking about is :)

GitjoPowershell, 2021-01-04:
So I misunderstood your point previously :)

Thanks for the clarification this is beyond my expectation

@cadriel 🙇 

I will now order fresh PEI 🤣

shinuza, 2021-01-07:
> 
> 
> Saving the adjustment during a print to your printer isn't possible, nor recommended. I think maybe taking note of the value during a print in order to apply once the print is complete is more feasible - which is really what the feature I was talking about is :)

I really think this should be a backend feature.

cadriel, 2021-01-10:
Im going to close this for now. Additional methods for adjusting z endstops etc will be coming in a future update - and are already monitored via a different list.

Thanks for the input all!

PR #489 (2022-02-06): Feat: Apply_Z_Offset
Resolves #57, Resolves #65
closed 2021-01-10 by cadriel