fluidd-core/fluidd #69: PiCam not visible on dashboard

tobor8man, 2021-01-04:
I just switched to Fluidd from Mainsail because I wanted to set up two Klipper printers on one Pi. I installed fluiddPi and uncommented the fluiddpi.txt parts for a raspicam (v2 cam on a Pi 3B+). After several updates/upgrades of OS/Moonraker/Klipper/Fluidd and reboots I have a working dashboard that can drive a printer but I see no camera output. However if I open a new browser tab and point to http://fluiddpi.local:8080/?action=stream the camera view is shown there. So the camera and mjpg_streamer setup seems to work. There are no errors in /var/logs/wecamd.log other than the warning about VL805 (Raspberry Pi 4). Not sure if I've done something wrong or there's a bug. I can poke around Linux a bit but I have no clue about http/browsers/json. Let me know if there's anything else I can do to help.

shinuza, 2021-01-04:
What value do you have in the settings?

[image]

tobor8man, 2021-01-04:
Duh ... sorry about that. Didn't even see those settings so it wasn't enabled. <embarrased>

closed 2021-01-04 by cadriel