fluidd-core/fluidd #73: [FeatureRequest] Hidden folders & no minimum character requirement

Kirinkunhime, 2021-01-05:
Good day!

I really enjoy fluidd, but it'd be nice to have a few things adapted / fixed in the jobs-overview:

- Have the ability to display hidden folders - e.g. I tend to keep some 'utility-prints' like bed leveling or other calibration tests in a .util-Folder. This causes it to always it to be displayed right at the top, even with a huge amount of other folders or files.
- Remove the minimum character requirement for folders. To be quite frank, I don't see what the issue is with it, because when SSHing into the Pi and creating the folder manually with 'mkdir' it can even be just 1 letter. And the folder can be navigated & used in fluidd without any issues.

Thank you for the time! :)

cadriel, 2021-01-07:
1. Hidden files and folders is due to other requirements in moonraker. For example, a `.thumbs` folder is created under certain circumstances. Also, other clients used to stick their json configurations in the `gcode` root. None of these items i'd want the average user from seeing. However, the filtering is a little heavy handed at the moment. I'll look into relaxing some of this.
2. Yeah, unsure why that's still around. I'll reduce it to 1.
3. Yep, should be doable.

Kirinkunhime, 2021-01-07:
Add 1.
I agree with you that per default hidden folders should not be visible. But as a toggle it'd make sense imho - because then then a user has to actively enable it & hopefully has a bit of knowledge what they're doing.

cadriel, 2021-01-08:
next release has relaxed folder name lengths.. 

the other points i'll comment here on if I get to them in time.

cadriel, 2021-03-15:
The next update will remove the restriction on hidden files, and there are no more length requirements on folders or files. This should now resolve this ticket. Please feel free to re-open after the next release if I've left something out.

closed 2021-03-15 by cadriel