fluidd-core/fluidd #74: Copying text out of the console puts separate outputs in reverse order

Taudris, 2021-01-05:
Sometimes I need to view more output text than the console can show, or I need to view the text and take actions based on it and need to see the text in a separate text editor window. (Mainly related to `DUMP_TMC` in my case.) Copying text from the console almost works, but multiple separate outputs are copied to the clipboard in reverse order. (Newlines within a single output are handled correctly, but Klipper uses a bunch of separate lines for `DUMP_TMC`.)

cadriel, 2021-01-06:
Yeah, the console is using a CSS trick to present the data RTL - the advantage being that it scrolls more naturally, the disadvantage being that copying the text from the console is a bit of a pain.

In order for me to fix this - I'd have to re-write how I handle the console output, which is on the list - but down in priority.

cadriel, 2021-01-16:
This will be resolved in the next release. Thanks!

closed 2021-01-16 by cadriel