fluidd-core/fluidd #75: min_extrude_temp value of 0 causes display issues in the Tool tab

FHeilmann, 2021-01-06:
my `min_extrude_temp` is 0, which causes this strange issue in the Tool tab:
[image]

cadriel, 2021-01-07:
Klipper gets angry at me when I set `min_extrude_temp` to 0. I assume you're on an older version of klipper for that to work. In any case, there was an issue. I'm pretty good at forgetting `0 === false` 👍🏼 

Anyway, I think I've fixed it - will be in next release. If I missed something, please re-open the bug.

closed 2021-01-07 by cadriel

FHeilmann, 2021-01-07:
I don't know why klipper gets angry at you. I'm on the latest version `v0.9.1-143` and I can set min_extrude_temp to 0 just fine. Thanks for fixing it :)