fluidd-core/fluidd #76: Fluidd Update not working

3derbauer, 2021-01-06:
I wanted to update the fluid client and now i´m waiting and waiting.   During the last update I had to restart the Pi and reinstall Fluidd. What can i do? 

[image]

cadriel, 2021-01-06:
Are you on a pi0?

cadriel, 2021-01-06:
this is likely a moonraker issue, if you could inspect or attach its logs when this happens then we can diagnose.

3derbauer, 2021-01-06:
P3B+
[moonraker.log](https://github.com/cadriel/fluidd/files/5778276/moonraker.log)

3derbauer, 2021-01-06:
Now i get a 403 message 
[image]

`2021/01/06 22:11:22 [error] 692#692: *52 rewrite or internal redirection cycle while internally redirecting to "/index.html", client: 192.168.10.102, server: _, request: "GET /config.json?date=1609967482279 HTTP/1.1", host: "fluiddpi.local", referrer: "http://fluiddpi.local/"
2021/01/06 22:11:25 [error] 692#692: *53 rewrite or internal redirection cycle while internally redirecting to "/index.html", client: 192.168.10.102, server: _, request: "GET /img/logo.6305d36d.svg HTTP/1.1", host: "fluiddpi.local", referrer: "http://fluiddpi.local/"
2021/01/06 22:35:25 [error] 692#692: *61 rewrite or internal redirection cycle while internally redirecting to "/index.html", client: 192.168.10.102, server: _, request: "GET /js/bedmesh.7f888f9b.js HTTP/1.1", host: "fluiddpi.local", referrer: "http://fluiddpi.local/"
2021/01/06 22:35:34 [error] 692#692: *62 directory index of "/home/pi/fluidd/" is forbidden, client: 192.168.10.102, server: _, request: "GET / HTTP/1.1", host: "fluiddpi.local"
2021/01/06 22:35:34 [error] 692#692: *62 rewrite or internal redirection cycle while internally redirecting to "/index.html", client: 192.168.10.102, server: _, request: "GET /favicon.ico HTTP/1.1", host: "fluiddpi.local", referrer: "http://fluiddpi.local/"
2021/01/06 22:35:38 [error] 692#692: *63 directory index of "/home/pi/fluidd/" is forbidden, client: 192.168.10.102, server: _, request: "GET / HTTP/1.1", host: "fluiddpi.local"
2021/01/06 22:35:42 [error] 692#692: *63 directory index of "/home/pi/fluidd/" is forbidden, client: 192.168.10.102, server: _, request: "GET / HTTP/1.1", host: "fluiddpi.local"
2021/01/06 22:35:46 [error] 692#692: *63 directory index of "/home/pi/fluidd/" is forbidden, client: 192.168.10.102, server: _, request: "GET / HTTP/1.1", host: "fluiddpi.local"
2021/01/06 22:35:51 [error] 692#692: *63 directory index of "/home/pi/fluidd/" is forbidden, client: 192.168.10.102, server: _, request: "GET / HTTP/1.1", host: "fluiddpi.local"
2021/01/06 22:36:17 [error] 692#692: *63 directory index of "/home/pi/fluidd/" is forbidden, client: 192.168.10.102, server: _, request: "GET / HTTP/1.1", host: "fluiddpi.local"
2021/01/06 22:36:36 [error] 692#692: *63 directory index of "/home/pi/fluidd/" is forbidden, client: 192.168.10.102, server: _, request: "GET / HTTP/1.1", host: "fluiddpi.local"
2021/01/06 22:36:52 [error] 692#692: *63 directory index of "/home/pi/fluidd/" is forbidden, client: 192.168.10.102, server: _, request: "GET / HTTP/1.1", host: "fluiddpi.local"
2021/01/06 22:38:14 [error] 717#717: *1 directory index of "/home/pi/fluidd/" is forbidden, client: 192.168.10.102, server: _, request: "GET / HTTP/1.1", host: "fluiddpi.local"
2021/01/06 22:39:52 [error] 717#717: *2 directory index of "/home/pi/fluidd/" is forbidden, client: 192.168.10.102, server: _, request: "GET / HTTP/1.1", host: "fluiddpi.local"
`

cadriel, 2021-01-07:
I think that error indicates your problem;

```
62 directory index of "/home/pi/fluidd/" is forbidden,
```

How did you install? Why is the fluidd folder forbidden to the the pi user? Have you made any changes yourself via SSH?

I'd start by double checking the permissions on the fluidd folder, and seeing if that helps.

3derbauer, 2021-01-07:
I have only installed the FluiddPI Image. No changes.

cadriel, 2021-01-08:
With your current installation, under the update menu - can you list the version for each component (or take a screenshot of it.. )

Arksine, 2021-01-08:
The Update appears to have failed due to a bug in Tornado.  I will look for workarounds.  With regard to the "forbidden", this indicates that either nginx lost permission to the Fluidd directory.   What is strange is that the update never made it to the point where client files were modified.

You will need to ssh into the Pi, I would try a reboot:  ```
sudo reboot now
```
If that doesn't solve it you may need to attempt to delete the files in the Fluidd directory and do a manual installation.

cadriel, 2021-01-14:
@3derbauer I'm going to close this issue. If you need further help, please feel free to reach out here or in Discord.

closed 2021-01-14 by cadriel