fluidd-core/fluidd #88: Klipper restart button prominent during page load

lilphil, 2021-01-09:
I just accidentally cancelled a print with the Restart button on mobile ui, as the restart button appears directly in the place of where the "force refresh" button was, for a split second during page load. I was trying to click the force refresh button as I thought something had gone funky, just as it changed.

I managed to grab a screenshot of what the interface looks like during load:

[image]

I'm not entirely sure what the fix would be for this, perhaps something to disable buttons until the UI finishes loading? An initial transparent overlay until it finishes loading? Maybe just a confirmation on the restart buttons?

I'm fairly new to fluidd, so perhaps everyone else has got used to ignoring the force refresh button, I was fortunate that the print was only half an hour in!

cadriel, 2021-01-10:
This is the Klipper state card. It shows up after Fluidd has connected to moonraker, but before it knows the state of your printer.

It shouldn't show for very long - but I suppose on slower connections it could show up for a bit longer. In any case, the only real way to not have it show would be to default Fluidd into assuming the printer is ok - and I'm not sure that's the best approach.

I'll be improving this state management moving forward, but for now it may have to be something you deal with.

closed 2021-01-10 by cadriel

cadriel, 2021-01-15:
I've found an issue that was delaying these from being removed from the screen quickly enough - this should be fixed in the next release!

Thanks for reporting!

lilphil, 2021-01-15:
Excellent, thanks!