fluidd-core/fluidd #92: Add double click home action

noobydp, 2021-01-12:
I keep finding myself clicking or double clicking the Home button (and I dont really like the hover over popup buttons).
Would you be able to add a double click action to the home button so double click homes all  axes?

Thanks :)

[image]

cadriel, 2021-01-14:
You're on an old version of Fluidd. Have you enabled the built in update manager?

Please upgrade.

closed 2021-01-14 by cadriel

noobydp, 2021-01-14:
Thanks!

I'm using the Kiuah install script and it doesnt look like it configured the update_manager section of moonraker.conf
I'll submit a request for them to add.