fluidd-core/fluidd #93: Replace online stylesheets with bundled one

expeditioneer, 2021-01-13:
Currently there are referenced online stylesheets in the index.html 

- https://cdn.jsdelivr.net/npm/@mdi/font@latest/css/materialdesignicons.min.css
- https://fonts.googleapis.com/css?family=Roboto:100,300,400,500,700,900"

It would be great when these could be replaced with bundled one. This could be integrated in the build process with something like https://www.npmjs.com/package/get-google-fonts.

The use case are printers or instances which have no internet connection.

cadriel, 2021-01-14:
Oversight on my part, thanks for spotting. I'll get this sorted in an upcoming release.

cadriel, 2021-01-16:
This will be in the next release. Thanks!

closed 2021-01-16 by cadriel