fluidd-core/fluidd #96: Feature Request: Interactive Console - Ability to click on ABORT, ACCEPT...

dimalo, 2021-01-15:
As a user I want to run through calibration routines quickly. In order to accelerate my calibration workflows I want to be able to click/tap on suggested choices during routines, such as ABORT, ADJUSTED, ACCEPT

Every time klipper prints some output with possible commands in it, they should be rendered as links/buttons/anything letting the user know he can click it.
Especially on the phone, having to type those commands is very time consuming.

I have tried using macros but they kind of clutter the macro selections and don't seem to be the right choice as it depends on the routine which commands are accepted and klipper tells you every time.

Very much appreciated 🥇

cadriel, 2021-01-16:
You're either a mind reader, or you're watching changes in develop.

This will be in the next release.

closed 2021-01-16 by cadriel

dimalo, 2021-01-16:
If I'd watched, I wouldn't have called out. Looking forward to it 👍