fluidd-core/fluidd #97: FR: RTSP camera support

dandantsui, 2021-01-15:
Forgive me if I am wrong but I've tried mjpgstreamer and IP camera stream type with a rtsp:// link and both options don't seem to load the rtsp stream.

Is there any way to get this working or does it needs a different stream type?

cadriel, 2021-01-16:
RTSP requires specialized components and code to support streaming in a web browser.

Instead, Fluidd supports the HTML 5 `<video>` tag. This gives native support for streaming protocols such as WebRTC.

Suggest looking into streaming something natively supported above, as that'd give you the best experience. `rtsp://` may come later if there's strong enough support, but for the moment - bigger fish to fry. 👍🏼

closed 2021-01-16 by cadriel

pedrolamas, 2022-11-07:
Right now, the best option for camera management is probably using [Crowsnest](https://github.com/mainsail-crew/crowsnest)!