fluidd-core/fluidd #103: FR: Don't disable the Jobs page when Klipper State is shutdown

LucidityCrash, 2021-01-18:
Being able to manipulate the files on the Virtual SD card while the MCU's are powered off (or any other reason that causes the Klipper State to be shutdown)  would be quite desirable for me.  Mainsail does this and apparently so does OctoPrint.

cadriel, 2021-01-18:
This is intentional for the moment.

Currently, fluidd (and mainsail) has no way of determining if the gcodes folder is working or not, until klippy has connected. So - for those people setting up their printers, the gcodes folder doesn't work - because moonraker hasn't connected to klippy - and this causes much confusion for people during setup.

On the other hand, those that power down their printer using the power plugin - CAN use gcode files - because klippy has previously connected.

Mainsail assumes gcodes is always ok - fluidd does not.

An upcoming change in moonraker will soon provide a way for us to determine the virtual_sdcard state, resolving this issue.

I'll keep this report open for now, until some kind of resolution is made.

cadriel, 2021-01-20:
This behaviour should now be changed for the next release.

If Fluidd knows that the gcodes folder exists, after a succesfull connection to klippy - then you should still be able to upload gcodes.

closed 2021-01-20 by cadriel