fluidd-core/fluidd #107: FR: Stop camera feed when panel is not visible

Taudris, 2021-01-20:
Currently, once the camera feed is started, it never stops until the page is reloaded. Things that should stop the feed include closing the Camera panel, and switching to a screen other than Dashboard.

My network connection to my Pi is sometimes not very good, so it can affect internet usage and UI responsiveness, not to mention it continues using CPU which is not great when on a laptop away from power. I don't really want to reduce camera quality or framerate since it's fine most of the time.

cadriel, 2021-01-20:
Are you on the latest copy of fluidd?

The DOM that loads the feed is already dropped when you collapse the camera panel, disable the camera panel, turn the webcam functionality off - or change pages.

Taudris, 2021-01-22:
I'm on the latest version as of a couple days ago. I can confirm the deletion of the DOM element. I guess it's a bug in Chromium?

I'm currently thinking a better solution will be [something like this.](https://community.octoprint.org/t/using-ffmpeg-for-webcam-streaming-and-timelapse-support/5321/16)

Taudris, 2021-01-22:
I found [a discussion about this issue](https://bugs.chromium.org/p/chromium/issues/detail?id=73395) which started back in 2011. It sounds like the solution/workaround is to set the src attribute to empty before removing the \<img> element. I was able to get this to work in the developer tools.

cadriel, 2021-01-22:
Re-opening this so I can verify and figure out if there's a fix my side or not.

cadriel, 2021-01-22:
@Taudris How are you confirming the stream is still running?

Taudris, 2021-01-22:
Network traffic in the Windows Task Manager. Less than 50kbits before
opening the camera, 10+ megabits after, and it persists until I reload the
page or set src to blank. I didn't specifically research how to see ongoing
traffic that has that x-replace-whatever mimetype in the dev tools, but I
didn't see anything obvious.

Supposedly using HTTP 1.1 also solves this problem (that info is in the
thread I linked), but mjpg-streamer doesn't have an active maintainer, the
changes required could be substantial, and the src workaround is
(seemingly) very simple. (I have no experience with Vue.)

cadriel, 2021-01-22:
Ok, I've added a change so that I set the url to nothing when you hide / show the camera panel - or navigate away.

We'll see how that effects things - will be in the next release. Let me know how you get on.

Taudris, 2021-01-24:
Hiding the camera panel using the toggle works and bandwidth usage drops to 0 as desired. Hiding the camera panel by switching tabs continues to consume bandwidth.

Interestingly, if I inspect the element in dev tools, collapsing no longer stops the bandwidth usage and a reload is required.

markwinger, 2021-01-31:
I notice when I open the UI when after the print is complete, the video is off and there is no way to start it.  I liked the fact that the video was always on with octoprint.  Then I could open the UI remotely and see if the print completed normally.  Not  big deal, just my preference.

cadriel, 2021-02-11:
@Taudris From what I'm reading, it sounds like there's not much else I can do here. I'm going to close this for now, but feel free to re-open if you feel like there's more we can do.

@markwinger I don't think this is related to this issue - but it may be fixed in the next release. If not, please create a new issue.

closed 2021-02-11 by cadriel

cadriel, 2021-02-12:
@Taudris Note - I have made another change here that will hopefully prevent the stream when the page not visible (I.e., you're on another tab).

Let me know how that goes for you.

Taudris, 2021-02-14:
It works well. Thank you!