fluidd-core/fluidd #108: Target temperature empty sends TARGER=NaN and causes the heater to heat to infinity

pszeder, 2021-01-20:
How to reproduce: 
- Delete any number from the Target temperature of the extruder
- Press enter
Result: 
- The contents of the Thermals will stop updating. (Maybe other things also stop updating, haven't checked). After entering a valid number it will start updating again. (Mainsail also stops updating the temperatures at the same time)
- Console says: $ SET_HEATER_TEMPERATURE HEATER=extruder TARGET=NaN
- Heater heats up to 270 degrees (my max temp) then printer Klipper shuts off. 

There could be some improvements that can be made in the UI, but my feeling is also that this is a Moonraker/Klipper problem as well, as NaN should not heat to infinity and beyond. I'll not post it there until I get a response here. 

I'm not too much into Klipper development (yet?), but if I get some encouragement I may look into it :)

For the UI, I would prefer to have a button next to each heater to disable them and I would parse 0 or NaN in the input as disabled.

cadriel, 2021-01-20:
[image]

I can't actually do what you suggest, but It's maybe because of a difference in `printer.cfg`. Do you have a `min_temp` and `max_temp` set for your extruder?

An upcoming change in klipper will likely fix this. Currently I don't get defaults for some values, but I need to know how you're reproducing in order to diagnose.

cadriel, 2021-01-20:
https://github.com/KevinOConnor/klipper/issues/3811

I've created an issue in Klipper for this also.

cadriel, 2021-01-20:
I'll also be adding a num check here to avoid this issue irrelevant of min / max's. Thanks for the report!

pszeder, 2021-01-20:
Thank you!

I had min_temp set as 0. After changing it to 1 and doing the same steps I've got invalid/red as you shown in your screenshot. This feels like that input validation is not applied in this case.

cadriel, 2021-01-20:
It wasn't checking validation when hitting return in this case.

Now resolved tho..

closed 2021-01-20 by cadriel