fluidd-core/fluidd #109: ui Settings are not individual in multiprinter configurations

klHaribo, 2021-01-20:
I configured 2 printers on one raspberry.
every printer has unique services and folders. everything configurable is renamed to the printers name.
Like ~/anycubic_config, ~/sparkcube_config (both have a individual .fluidd.json)
The problem seems to effect the whole UIsettings. If I rename one printer, the other gets renamed too. Same with camera-url (http://192.168.1.**10**:8080/?action=stream vs http://192.168.1.**11**:8080/?action=stream

PS: fluidd is still awesome, really great work :)

cadriel, 2021-01-20:
Are you saying you have multiple printers running on one host / moonraker setup?

Currently - UI settings are saved PER printer PER moonraker instance. The settings are saved to `.fluidd.json` via moonraker, on the instance you're currently working with.

How are your printers setup / configured?

klHaribo, 2021-01-21:
sorry i had not time until now :/

this is my config(changed parts):

**klipper:**
/etc/init.d/anycubic
	DEFAULTS_FILE=/etc/default/anycubic
	PIDFILE=/var/run/anycubic.pid

/etc/init.d/sparkcube
	DEFAULTS_FILE=/etc/default/sparkcube
	PIDFILE=/var/run/sparkcube.pid

/etc/default/anycubic
	KLIPPY_ARGS="/home/pi/klipper/klippy/klippy.py /home/pi/anycubic_config/anycubic.cfg -l /tmp/anycubic.log -a /tmp/anycubic_uds -I /tmp/anycubic"

/etc/default/sparkcube
	KLIPPY_ARGS="/home/pi/klipper/klippy/klippy.py /home/pi/sparkcube_config/sparkcube.cfg -l /tmp/sparkcube.log -a /tmp/sparkcube_uds -I /tmp/sparkcube"

**moonraker:**
/etc/systemd/system/moonraker-anycubic.service
	ExecStart=/home/pi/moonraker-env/bin/python /home/pi/moonraker/moonraker/moonraker.py -c /home/pi/anycubic_config/moonraker.conf

/etc/systemd/system/moonraker-sparkcube.service
	ExecStart=/home/pi/moonraker-env/bin/python /home/pi/moonraker/moonraker/moonraker.py -c /home/pi/sparkcube_config/moonraker.conf

~/anycubic_config/moonraker.conf
	[server]
	host: 0.0.0.0
	port: 7125
	enable_debug_logging: True
	config_path: ~/anycubic_config
	klippy_uds_address: /tmp/anycubic_uds

	[authorization]
	enabled: True
	cors_domains:
	    http://anycubic.local
	    http://app.fluidd.xyz

~/sparkcube_config/moonraker.conf
	server]
	host: 0.0.0.0
	port: 7126
	enable_debug_logging: True
	config_path: ~/anycubic_config
	klippy_uds_address: /tmp/sparkcube_uds

	[authorization]
	enabled: True
	cors_domains:
	    http://sparkcube.local
	    http://app.fluidd.xyz

klHaribo, 2021-01-21:
oh!
I found the mistake!

closed 2021-01-21 by klHaribo

cadriel, 2021-01-21:
haha, phew - you had me worried! :P

emersonsc, 2022-01-02:
> oh! I found the mistake!

What was the mistake? I am currently looking to convert another one of my printers to Klipper, and want to use the same PI to control both of my Klipper printers since I have a nice 7" touch screen i use to control it, that and its a Pi 4B and even at its heaviest, my current printer maybe hits 20% load.