fluidd-core/fluidd #111: Show heater control parameters and fans in graphs

dimalo, 2021-01-23:
As a user calibrating his printer even to the PID parameters, I would like to see the heaters' control parameters in the UI.

I remember I have seen values like heater power (0-1.0) in the status message payload, pretty sure P, I, and D parts are not propagated :(

It would be very helpful seeing at least the heater power over time in correlation to the temperatures so an oscillating control can be easily spotted. Maybe having a multiselect to choose which values to show in the graph? That way more can be added, also with adding heaters to the setup...

Same is valid for fans, as they tend to influence the temperatures shown.
That way users can spot lack of power in fully powered fan situations like bridges...

In my opinion auto tuning PID is a very rudimental approach for non-linear systems like heaters.
Especially on heated bed PID control, only applying auto tuned parameters causes the power output to oscillate a lot on steady setpoints. Having a way of showing this to the users in my opinion would surface the lack of control accuracy nicely!

dimalo, 2021-01-24:
@cadriel brilliant to see power already being displayed in 1.6! thank you!

cadriel, 2021-02-11:
This will be in the next release. PID values and temperature fans will also be shown..

Please let me know if there's something I've missed here. Feel free to re-open if required after the next release.

closed 2021-02-11 by cadriel