fluidd-core/fluidd #115: Docker images don't have release tags

dimalo, 2021-01-24:
[https://hub.docker.com/r/cadriel/fluidd/tags](https://hub.docker.com/r/cadriel/fluidd/tags)

just showing latest only, even on building only for releases.

perhaps triggering by the release workflow being completed doesn't pass tags compared to trigger on push?

cadriel, 2021-01-24:
Possibly, the problem is that the docker publish can't happen until after the release, otherwise it'll pick up the wrong release artifact.

It really shouldn't be otherwise built / released outside of an actual Fluidd release.

I am aware of the issue - but haven't had time to investigate a solution yet.

cadriel, 2021-01-24:
This should now be resolved

closed 2021-01-24 by cadriel

dimalo, 2021-01-24:
confirming - great job!