fluidd-core/fluidd #118: Docker: warning about misconfigured update_manager

dimalo, 2021-01-25:
[image]

Just having set

[update_manager]

in moonraker.conf

Does fluidd expect client update_manager entries? I am using the default config.json

dimalo, 2021-01-25:
Printer and moonraker are working fine though
Will have a look into the moonraker logs...

cadriel, 2021-01-25:
This is not unique to Docker. It's telling you that one of your moonraker plugins is failing.

If you're using the latest docker (v1.6.1) then you need to make sure Klipper and Moonraker are up to date.

closed 2021-01-25 by cadriel

dimalo, 2021-01-25:
They are up to date. Not sure why you're closing this - the issue is still there. I can change the title though...

dimalo, 2021-01-25:
finally had the time to go through logs. yep a moonraker issue...

```confighelper.ConfigError: Invalid path for option 'python_dist_path'```