fluidd-core/fluidd #119: Accurate Printtime calculation

nicedevil007, 2021-01-26:
Heya, 

Just want to ask if there will be at any time a good solution for accurate printtimes like it is not n Octoprint with printtime genius (https://github.com/eyal0/OctoPrint-PrintTimeGenius)

After upload it took sone time for the calculation and then it was the most accurate timeleft calculations I've ever had since I'm 3D printing.

Maybe you can implement this in your code as well? Or ofc any similar solution?

Taudris, 2021-01-27:
[Relevant Klipper issue.](https://github.com/KevinOConnor/klipper/pull/890)

cadriel, 2021-02-07:
There's discussion happening surrounding print times in klipper / moonraker. I'm going to close this for now as it's dependent on these.

Thanks!

closed 2021-02-07 by cadriel