fluidd-core/fluidd #121: Print time / % keep intermittently showing and disappears

Nandox7, 2021-01-28:
Hi,

I noticed that the stats for elapsed and remaining time keep showing and disappearing during the print.
Only the other values keep showing all the time.

[image]

Don't know exactly who is responsible for that ata (Moonraker?) but I also have a LCD with KlipperScreen and the values
show all the time during the print.

cadriel, 2021-01-28:
What estimates setting are you using?

Nandox7, 2021-01-28:
I'm using "File estimation".

cadriel, 2021-02-07:
I believe this is now resolved by way of the 1.6.3 commit. Please feel free to re-open if that is not the case.

closed 2021-02-07 by cadriel