fluidd-core/fluidd #123: Wrong gcode prints due to case sensitivity

BeansIsFat, 2021-01-28:
If a gcode already exists in Jobs, uploading a new gcode with the same name but different capitalization can result in the old gcode being printed.

Steps to reproduce:
1. Place my_print.gcode in Jobs.
2. Upload a new print from slicer
3. Name print My_print.gcode 
4. Check the box "Start printing after upload"

Result:
My_print.gcode will upload but my_print.gcode will start printing.

andrepl, 2021-02-25:
Unless I'm mistaken, this is an issue with Klipper, not fluidd.

https://github.com/KevinOConnor/klipper/blob/d9af6129eacacefa6b33d6ed45a593ef6e11fddb/klippy/extras/virtual_sdcard.py#L113

wlhlm, 2021-02-26:
Upstream PR: https://github.com/KevinOConnor/klipper/pull/3994

Thanks @andrepl

wlhlm, 2021-02-27:
Upstream fix has been merged. This issue can be closed now.

closed 2021-03-01 by BeansIsFat