fluidd-core/fluidd #124: macros don't get executed if named in special way

Spacemarine2018, 2021-01-29:
I needed a simple button, that moves the x-axis to 200. So I wrote:

[gcode_macro g1_x200]
gcode:
  g1 x200

However, this macro doesn't do anything and also shows no error. I can push the button, but nothing happens.
After I renamed it, it works:

[gcode_macro x200]
gcode:
  g1 x200

cadriel, 2021-01-30:
Are you sure this is a fluidd issue? Did you try to run the g1_x200 macro from the console to confirm it worked there first?

markwinger, 2021-01-30:
This is klipper.  The docs specify that the names can only have numbers at the end, not embedded in the middle.

Spacemarine2018, 2021-01-31:
Sorry, this is my fault then, I wasn't aware of that restiction. However, I think it would be good for klipper to raise a warning in that case, but that's another problem.

Maybe fluidd can refuse to create macro buttons with wrong formatting?

cadriel, 2021-01-31:
Really, it's up to moonraker / klipper to validate these. Fluidd only represents the data passed back to it from the API.

closed 2021-01-31 by cadriel