fluidd-core/fluidd #127: Feature Request

markwinger, 2021-01-30:
I just installed fluidd yesterday (was using octoprint) and I lilke it so far.   It does not have all the configrurability and plugins like octoprint, but I do like fluidd and plan on staying with it.

I created several macros, (it took some experimenting) and got them working.  But the only things UI wise I can do with them is turn them on or off with UI settings, and re-order them by reordering the macros in printer.cfg.  I would like to be able organize them a little more.

For example, lets say I have several bltouch macros,  and several for setting z-offset.  Right now they just appear as a bunch of buttons.

Maybe having a macro that does nothing, but creates a title, or causes all buttons that follow to be on a following line.  It seems like you could create blank macros and use the macro name to control layout.  If a macro was named BLOCK_BLTOUCH, you could create the title line BLTOUCH.  Then following macros be below that  until the next macro BLOCK_ZOFFSET appears.  That could create a new title line ZOFFSET, an following buttons would be below.  

Then you could could group buttons with separating group names.  Maybe even add outlines around them.  Just suggesting what seems easy to do it without messing up klipper configuration.

cadriel, 2021-01-31:
Ideas surrounding macro configuration are on the roadmap. Will leave this here for brevities sake for now.

JQCoburn, 2021-02-05:
I had a couple requests related to macro features/improvements:
- I'd love to be able to hide certain macros. I have custom gcode macros like M900  to translate slicer ouput to klipper, but I don't need a button cluttering up the UI for it. 
- Some macros need input variables to useful, it'd be cool if the UI supported setting those parameters.

cadriel, 2021-02-05:
@JQCoburn Go to the UI settings page, you can already hide macros.

JQCoburn, 2021-02-07:
@cadriel Thanks! I didn't realize. :)

cadriel, 2021-05-07:
Hi! I'm closing this, the latest release has further macro enhancements, and soon to come will be the ability to also order them.

If you have further comment, or want to provide more feedback - please feel free to comment here or join the Discord!

https://discord.gg/GZ3D5tqfcF

closed 2021-05-07 by cadriel