fluidd-core/fluidd #128: Temp presets oftena ren't listed, required few forced page refresh to show them again

Nandox7, 2021-01-31:
Hi,

Logging this for tracking, not a major issue but annoying.
Often and it can be at any time I do not see my temp presets listed in dashboard, usually requires a few page refresh to have them lifted again.

[image]

3x Page refresh later
[image]

cadriel, 2021-01-31:
I've not been able to reproduce this. Can you reliably provide a way to reproduce it?

The only way in theory this would happen is if your configuration can't read from moonraker.

Nandox7, 2021-01-31:
Yes I can, have the machine printing and just tried and and empty list.
Did a page refresh and the list was populated.

cadriel, 2021-02-01:
No, this data gets read when you load the UI - via an XHR GET request for `fluidd.json`. Then, any time you change a setting on the UI settings page, it gets rewritten via another XHR POST.

If you examine dev tools in chrome, you might see an error. Otherwise - I can't see how to reproduce this. I'd expect other UI settings to not stick either if that XHR call was broken in some way.

Have you ensured you're on the latest releases of Klipper, Moonraker and Fluidd?

cadriel, 2021-02-10:
Can you re-check to see if this is still occurring? The 1.6.3 release may have fixed it.. if not, i'd like to close this issue as I can't reproduce.

nicedevil007, 2021-02-11:
@Nandox7 did you ever tried a whole reboot of your Pi and then cleared the cache of your browser? I got this as well on a past release and it was gone after rebooting the pi and cache clearing of my browser.

Nandox7, 2021-02-11:
Apologies as I meant to update this here before.
I got to catch  it and it seems to happen on the first load of the page (as I don't use it all the time probably I'd noticed it after some time).
[image]
I caught it in debug mode but is a bad screenshot, still I can see that it fails to load the .json file.

My suspicion is that it may be due to the DNS lookup failing.

@nicedevil007, reboot yes multiple times, not on purpose but I'd occasionally reboot the pi. At the moment I actually had to re-install all the full image so I'll try this again.

cadriel, 2021-02-12:
What's the URL you type into your browser to access Fluidd?

Are you able to upload gcode files ok?

That error is normal when you first hit Fluidd, before you've changed any UI settings. After you've defined a setting, you'd see a similar call that would attempt to write the file (versus reading it, as above).

Try keeping that dev tool open - and changing something in UI settings. That would better indicate if it's having trouble also writing the file.

Furthermore, seeing more detail as to WHY the request failed would be helpful. Sometimes the line item before you click into the failed file can garner more information too.. so maybe paste that.

That API call is direct to moonraker, and if its an intermittent issue reading that file (and the call doesn't return a status) then I'd hazard a guess it might be something to do with networking. It might be a good idea to plug an ethernet cable in temporarily to see if that resolves the issue, it'd at least help diagnose.

cadriel, 2021-02-23:
I'm going to close this due to inactivity - feel free to re-open should it still be an issue.

closed 2021-02-23 by cadriel