fluidd-core/fluidd #129: push/browser Notification

imrcly, 2021-01-31:
Browser notification when a print completes/*error stops*

Tried to find this in vueify for desktop browser but did find it as an html standard https://developer.mozilla.org/en-US/docs/Web/API/Notification
and as a  push notification view for vue on mobile devices https://vuejsexamples.com/a-library-for-previewing-push-notification-with-vue/

cadriel, 2021-01-31:
I've been hesitant to implement this in the past, due to a lot of users usually blocking browser based notifications - or them not being well integrated with the OS.

jtkDvlp, 2021-02-01:
It would be nice to get notifications on mobile via **P**rogressive **W**eb **A**pp design

cadriel, 2021-02-02:
Fluidd is already a PWA. The notifications themselves are another story.. I'll look into them a bit more in an upcoming update.

jtkDvlp, 2021-02-02:
If desired please give a little developer overivew / intro documentation and I offer my help :-)

matmen, 2022-03-14:
I've started implementing this, but we are running into a couple issues here:
- [Chrome](https://groups.google.com/a/chromium.org/g/Blink-dev/c/IVgkxkRNtMo) and [Firefox](https://bugzilla.mozilla.org/show_bug.cgi?id=1429432) no longer allow notifications to be instantiated from insecure origins (meaning only HTTPS is supported)
- To support secure origins, Moonraker will have to be served over HTTPS too to prevent mixed content
- I believe audio cues (as suggested in #520) also won't work because playing sound requires user interaction

Because of those reasons, deploying this is currently blocked (as it would only work for localhost or setups with both Fluidd and Moonraker being served over HTTPS). I'll move this to the backlog and keep you updated if anything changes.

pedrolamas, 2022-07-18:
Given the technical requirements for in-browser notifications to work (specifically, HTTPS), I don't see this moving forward anytime soon.

I would recommend that instead, users just use the existing Moonraker notifier support: https://moonraker.readthedocs.io/en/latest/configuration/#notifier

closed 2022-07-19 by matmen