fluidd-core/fluidd #130: Current layer value

Swift-Tester, 2021-02-03:
Hello!

Please consider adding currently printing layer value.
Maybe right after "Z" in "Tool" widget.

Thanks in advance.

cadriel, 2021-02-07:
There's no way to reliably do this because klipper and moonraker don't give us this value. Working it out based on layer height and total object height is unreliable due to prints being able to have many layer heights.

If this comes in klipper side, I'll implement it.

Thanks!

closed 2021-02-07 by cadriel

maruhe, 2021-06-20:
how does mailsail know the layer and count?

[image]

for me this is the only missing information from "Mainsail printing-widget" and I think I can stop using mailsail as fluidd shows the current layer-number and total count :-*

custom buttons in header would be nice also ;-)

cadriel, 2021-06-20:
Mainsail assumes your prints don't have variable layer heights. :)

If you print a variable layer height print, then those values won't be true. In any case, this will be coming in a future update of Fluidd.

maruhe, 2021-06-23:
so layer-indicators in gcode-file get ignored by klipper/moonraker?
this would have to be needet to be implemented in klipper first?
the currenlty shown layer-nr in fluidd also fit's only without variable layer height?

cadriel, 2021-06-23:
Klipper doesn't parse a gcode file up front. In order to get a true layer count, it'd need to read in the entire file file - this is not how Klipper works. 

But otherwise yeah, what you've said is true.. its our best guess without taking into account variable layer heights.