fluidd-core/fluidd #131: Remote Access

markwinger, 2021-02-04:
I have port forwarding setup that I used with OctoPrint but am unable to make this work with fluidd.

Octoprint has a login page, fluidd does not.  Not sure it that is why.  But when i connect via lan it just opens fine.  But when I connect via wan fluid opens but give me the add printer dialog.  If I cancel, it says connection to printer and never connects.  Clilcking force refresh goes to add printer again.  

If I try to configure a printer it does not work.  Should this work?

cnf, 2021-02-06:
I think you need to set up moonraker authorization as per https://github.com/Arksine/moonraker/blob/master/docs/configuration.md#authorization and https://github.com/Arksine/moonraker/blob/master/docs/web_api.md#authorization

Some more AAA might be nice, but Fluidd is just a front-end for moonraker, so all AAA implementations would need to happen in moonraker.

markwinger, 2021-02-06:
I had looked at that but was unclear how to set that up.  If attempting to connect via phone over verizon, I assume the ip address will change (no static ip), meaning using trusted client wont work.  

So that would mean using a api-key.  The config file says:
#   Enables authorization.  When set to true, requests must either contain
#   a valid API key or originate from a trusted client. Default is True.
api_key_file: ~/.moonraker_api_key
#   Path of the file that stores Moonraker's API key.  The default is
#   ~/.moonraker_api_key

I see the .moonraker_api_key file, but how do I get the browser to persent the api_key?

cnf, 2021-02-06:
> I had looked at that but was unclear how to set that up. If attempting to connect via phone over verizon, I assume the ip address will change (no static ip), meaning using trusted client wont work.

That is correct
> ...
> 
> I see the .moonraker_api_key file, I tried adding ?key={API_KEY} to the url but that did not seem to work. I still get the add printer dialog when trying it. I think I'll try asking this on moonraker.

From the moonraker page: `Untrusted Clients must use a key to access the API by including it in the X-Api-Key header for each HTTP Request. The API below allows authorized clients to receive and change the current API Key.`

Try adding X-Api-Key instead of just key

markwinger, 2021-02-06:
?X-Api-Key={key}
Same Result.

Nothing shows up in the moonraker logs when I try to connect, but I still get the add_printer dialog.  I thought moonraker would at least log the attempt to connect.  When I connect locally I see " Trusted Connection Detected, IP:....".

cadriel, 2021-02-07:
We have plans to implement a more solid solution surrounding this, with 1st class authorization within Fluidd. However, there's a bunch of complication surrounding this as it requires HTTPS on the client, which isn't something the average user should need to think about.

So - for the moment, auth is a manual process users need to figure out via VPN or similar until we've got a better solution in place.

Denny86, 2021-02-07:
Use the v1.5.1, i have the same problem with v.1.6.1.

markwinger, 2021-02-08:
I'm on V1.6.2.  Are you saying it worked at V1.5.1?

cadriel, 2021-02-10:
I'm going to close this issue as it's not directly related to Fluidd. Best place to work this out would be the various discords;
https://discord.gg/GZ3D5tqfcF

Thanks!

closed 2021-02-10 by cadriel