fluidd-core/fluidd #136: Button JOBS is gray.

anjoku, 2021-02-06:
hello, i installed fluidd with the help of KIAUH.

I am using an Orange Pi Lite. Controlling the printer works. But I can't load files. My Jobs button is gray and I can't choose it.

What am I doing wrong?

Many thanks
Andre
[image]
[klippy.log](https://github.com/cadriel/fluidd/files/5936566/klippy.log)
[moonraker.log](https://github.com/cadriel/fluidd/files/5936567/moonraker.log)

anjoku, 2021-02-07:
I installed it again yesterday. Armbian.
The following order:
User Andre

1. Armbian and upgrade
2. KIAUH
From KIAUH
1. Klipper
2. Moonraker
3. Fluid

The printer can be controlled, but Jobs is still not accessible

cadriel, 2021-02-07:
jobs is only available once moonraker / klipper tell fluidd it is. Are you getting other warnings or errors in fluidd? The issue you're describing suggests your virtual_sdcard is not configured properly or your printer isn't connected.

cadriel, 2021-02-07:
Also, you can ask in the various fluidd / klipper channels in discord for further help on this.

closed 2021-02-07 by cadriel