fluidd-core/fluidd #138: Heater Fans Display in a strange manner in the UI

adracea, 2021-02-06:
Heater fans such as the cooling block fan seem to be displaying oddly. There shouldn't be anything controllable about them so this is more of an aesthetic issue : 

[image]

adracea, 2021-02-07:
To add something more constructive :
The way it looks now is a bit confusing and could probably do with a bit more emphasis on the fact that it's not controllable and that the status is at 100%. Maybe display it like : 
```
Heater Fan ........... Status
Fan_Name ............ On/Off
```
In order to make it look more aligned with the part fans ?

cadriel, 2021-02-11:
Ok, so I've had some thoughts here.

The problem is that fan's can be configured to not be at 100%. I.e., my hotend fan is configured to hit 65% maximum.

Also - you can configure fans to be PWM'd based on other events; So the display has to show that the fan is on, and at what power the fan is running at.

cadriel, 2021-02-11:
As an interim solution;

```
Fan_Name ............. Off
Fan_Name ............. 65%
Fan_Name ............. 100%
```

cadriel, 2021-02-11:
Feel free to add some more input / re-open this if required.. the above change is in for the next release as an initial change.

closed 2021-02-11 by cadriel