fluidd-core/fluidd #144: Changing flow rate causes // Unknown command:"M205"

sajrashid, 2021-02-12:
Hi,

**Fluidd** v1.6.3
**Klipper** v0.9.1-222
**MoonRaker** v0.3.0-5

Issue is the same from the _UI_ or from the _console_ when sending M221 

**from Klippy log**

Stats 3299.0: gcodein=0 mcu: mcu_awake=0.022 mcu_task_avg=0.000023 mcu_task_stddev=0.000025 bytes_write=4596705 bytes_read=709738 bytes_retransmit=9 bytes_invalid=0 send_seq=78714 receive_seq=78714 retransmit_seq=2 srtt=0.001 rttvar=0.002 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=99996242 host: mcu_awake=0.001 mcu_task_avg=0.000013 mcu_task_stddev=0.000012 bytes_write=17951 bytes_read=57075 bytes_retransmit=0 bytes_invalid=0 send_seq=2972 receive_seq=2972 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=49999413 adj=50001257 sd_pos=631176 heater_bed: target=55 temp=55.0 pwm=0.093 print_time=3269.458 buffer_time=2.117 print_stall=1 extruder: target=235 temp=235.0 pwm=0.580 sysload=0.98 cputime=77.765 memavail=3392048
Unknown command:"M205"
Unknown command:"M205"
Unknown command:"M205"

repetatively

cadriel, 2021-02-13:
This is not a Fluidd issue. If you're seeing invalid gcode commands in your console / fluidd - it's likely a slicer or configuration issue.

`M221` is a valid klipper gcode, `M205` is not. Have you configured your slicer?

sajrashid, 2021-02-13:
Hi, makes sense, I'm using Cura which is configured for Marlin.
I'll try SuperSlicer, will update later today.

thanks for getting back to me & thank you for this wonderful UI

closed 2021-02-13 by cadriel

sajrashid, 2021-02-15:
tested with SS and manually all good, thanks