fluidd-core/fluidd #145: Cannot Access fluid from multiple computers

Custom3DProducts, 2021-02-13:
When trying to access fluid from other computers on the network it only brings up the add printer screen. Putting in http://fluidpi.local also doesnt work. Works fine from my other computer. 
[image]

cadriel, 2021-02-13:
Please see the documentation here;
https://github.com/cadriel/fluidd/blob/develop/docs/setup/moonraker.md

Or join the Discord for help. In your case, you need to setup `cors_domains` or access fluidd by its host.

closed 2021-02-13 by cadriel

Custom3DProducts, 2021-02-13:
Thank you! 
I added 
`trusted_clients:
 10.8.121.0/24`
and 
`cors_domains:
  http://10.8.121.103:80`
works perfectly now!