fluidd-core/fluidd #147: Performance: All console history entries are re-rendered

Piezoid, 2021-02-13:
When the console history is full (1000 entries) new entries are displayed very slowly and the frame time reach ~1s on Chrome and Firefox. This is most obvious with big outputs (`DUMP_TMC` or tuning tower with a spiralized contour)
[Profile screeshot](https://user-images.githubusercontent.com/3748582/107860965-fc04c500-6e42-11eb-96cd-082811c53da1.png)

Reducing `config.CONSOLE_HISTORY_RETENTION` to 100 solved the problem, confirming the O(n) render time theory.
The DOM view shows that all entries and their container gets updated (pink flash) each time a new one is added.

I've tried to key the `ConsoleEntryWidget`s `v-for` with a monotonic key, instead place of the dequeue's indices, to help the diffing algorithm avoid shifting the content of all entries. This allows to use `v-once` inside console entry, which was not working before because of the content shift. It helped a bit (frame time dropped to 400ms), but still very slow.
[Profile screeshot](https://user-images.githubusercontent.com/3748582/107860812-f5298280-6e41-11eb-8d3a-0de12643112b.png)

Possibly, implementing a virtual scrolling would help the most, but not being able to avoid rendering of all entries is cumbersome.

cadriel, 2021-02-14:
Yep. This is likely due to implementation and the changes recently to do with filtering, and command clicks.

Really the data should be appended once without the need for additional rendering on already appended items.

I'll take a look at making some adjustments here for an upcoming release.

cadriel, 2021-02-14:
Ok, so there was some nasty things going on here;

1. Items were re-rendering as the array changed. This should no longer be happening, due to items now having an actual unique key over and above their index.
2. Some slowdown was also introduced by slicing the array on the watcher. I was doing this because I had to determine if there was a change to the array in order to determine if I needed to scroll to the bottom or not. This is still happening now, but i'm doing a shallow clone in the getter instead. The array still causes a re-render of items if you change the filter applied to the data (i.e., toggle the temp waits option), which leads me to the last point;
3. We need a virtual scroll. This will prevent the wait times associated with having to re-render 1000+ dom elements (or having to remove them.

The issue isn't having 1000 items in the DOM, but having to remove / add them IS. I have steps 1 and 2 sorted, and am working on a solution for 3.

cadriel, 2021-02-14:
@Piezoid 

I have a fix in place. Would you be able to run your tests on the attached test build? If you come up with no issues - I'll merge. Thanks!
[fluidd.zip](https://github.com/cadriel/fluidd/files/5978524/fluidd.zip)

Piezoid, 2021-02-14:
> I have a fix in place. Would you be able to run your tests on the attached test build? If you come up with no issues - I'll merge. 

Great, thanks! I can confirm that it's much more fluid now, with the virtual scrolling.

> The issue isn't having 1000 items in the DOM, but having to remove / add them IS.

Yes, the issue was updating/rendering all of them for inserting a single entry.

I think 1. and 3. were the most important improvements, ad they reduce the size of DOM updates which was the point of friction. I was a bit concerned about the added complexity and dependencies of virtual scrolling, but since you went for it, I guess it's fine.

One potential improvement would be to batch multiple insertions in one update. Currently, it seems that only a single entry can be added per DOM update.

closed 2021-02-15 by Piezoid