fluidd-core/fluidd #155: Camera URL "cacheBust" is not valid parameter

kenblu24, 2021-02-17:
I just updated to v1.7.1 from something like v1.6.9? I forget what version I had before. Camera embedding is not working for me anymore, whereas it previously did. 

It seems fluidd is adding an anti-cache parameter to the URL to ensure that a stale image is not loaded. However, my webcam supports a live stream via the file `/video.cgi`. Something about the added URL parameter is causing it to not load for me. 

[image]

Here's what does not load for me:

`http://sub.test.net:48000/video.cgi=&cacheBust=1613595163086`

The above URL will give me a File Not Found error.

Here's what does load for me: 

`http://sub.test.net:48000/video.cgi`**`?`**`=&cacheBust=1613595163086`

`http://sub.test.net:48000/video.cgi?cacheBust=1613595163086`

cadriel, 2021-02-18:
Yep, that's an incorrectly formatted URL.

I'll get a fix in for the next release, sorry about that.

cadriel, 2021-02-20:
This is sorted in code, please re-open if this is not resolved with the next release. Thanks for reporting!

closed 2021-02-20 by cadriel