fluidd-core/fluidd #156: Interface broken.. JOBS grayed out.

Anton33455, 2021-02-21:
My interface is broken  "Jobs" menu item is grayed out.  tried reinstalling, no luck.  Installed all latest updates
Fluid v1.8.1  Klipper v0.9.1-256  Moonraker v0.3.1-6. 

Any ideas what would cause this?

Anton33455, 2021-02-21:
[image]

Anton33455, 2021-02-21:
# This file contains common pin mappings for the BIGTREETECH SKR V1.3
# board. To use this config, the firmware should be compiled for the
# LPC1768.

[stepper_x]
step_pin: P2.2
dir_pin: !P2.6
enable_pin: !P2.1
step_distance: .0125
endstop_pin: P1.29  # P1.28 for X-max
position_endstop: 0
position_max: 200
homing_speed: 100

[stepper_y]
step_pin: P0.19
dir_pin: !P0.20
enable_pin: !P2.8
step_distance: .0125
endstop_pin: P1.27  # P1.26 for Y-max
position_endstop: 0
position_max: 200
homing_speed: 100

[stepper_z]
step_pin: P0.22
dir_pin: P2.11
enable_pin: !P0.21
step_distance: .00125
endstop_pin: !P1.25  # P1.24 for Z-max
position_endstop: 0.5
position_max: 200

[extruder]
step_pin: P2.13
dir_pin: P0.11
enable_pin: !P2.12
step_distance: .0012195
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: P2.7
sensor_type: ATC Semitec 104GT-2
sensor_pin: P0.24
#control: pid
#pid_Kp: 30.642
#pid_Ki: 1.318
#pid_Kd: 178.104
min_temp: 0
max_temp: 285

#[extruder1]
#step_pin: P0.1
#dir_pin: P0.0
#enable_pin: !P0.10
#heater_pin: P2.4
#sensor_pin: P0.25
#...

[heater_bed]
heater_pin: P2.5
sensor_type: EPCOS 100K B57560G104F
sensor_pin: P0.23
#control: watermark
#control: pid
#pid_Kp: 70.939
#pid_Ki: 2.642
#pid_Kd: 476.178
min_temp: 0
max_temp: 130

[fan]
pin: P2.3

[mcu]
serial: /dev/serial/by-id/usb-Klipper_lpc1768_03000014A78C3CAF29DFFD5CC42000F5-if00

[virtual_sdcard]
path: ~/sdcard

[pause_resume]
[display_status]

[gcode_macro CANCEL_PRINT]
rename_existing: BASE_CANCEL_PRINT
gcode:
    TURN_OFF_HEATERS
    CLEAR_PAUSE
    SDCARD_RESET_FILE
    BASE_CANCEL_PRINT

[gcode_macro front_left]    
gcode:
     g28
     G1 Z5 F200
     G1 X10 Y10 F2000
     G1 Z0 F100
     
[gcode_macro front_right]    
gcode:
     g28
     G1 Z5 F200
     G1 X200 Y10 F2000
     G1 Z0 F100
     
[gcode_macro back_left]    
gcode:
     g28
     G1 Z5 F200
     G1 X10 Y180 F2000
     G1 Z0 F100    
[gcode_macro back_right]    
gcode:
     g28
     G1 Z5 F200
     G1 X180 Y180 F2000
     G1 Z0 F100

[printer]
kinematics: cartesian
max_velocity: 200
max_accel: 2000
max_z_velocity: 25
max_z_accel: 100

#### end of menu

########################################
# TMC2208 configuration
########################################

# For TMC2208 UART
#   1) Remove all of the jumpers below the stepper drivers
#   2) Place jumpers on the red pin headers labeled XUART (XUART, YUART etc.)

[tmc2208 stepper_x]
uart_pin: P1.17
microsteps: 16
run_current: 0.800
hold_current: 0.500
stealthchop_threshold: 250

[tmc2208 stepper_y]
uart_pin: P1.15
microsteps: 16
run_current: 0.800
hold_current: 0.500
stealthchop_threshold: 250

[tmc2208 stepper_z]
uart_pin: P1.10
microsteps: 16
run_current: 0.850
hold_current: 0.650
stealthchop_threshold: 30

#[tmc2208 extruder]
#uart_pin: P1.8
#microsteps: 16
#run_current: 0.800
#hold_current: 0.500
#stealthchop_threshold: 5

#[tmc2208 extruder1]
#uart_pin: P1.1
#microsteps: 16
#run_current: 0.800
#hold_current: 0.500
#stealthchop_threshold: 5

########################################
# TMC2130 configuration
########################################

# For TMC SPI
#   1) Place jumpers on all the red pin headers under the stepper drivers
#   2) Remove jumpers from the red pin headers labeled XUART (XUART, YUART etc.)

#[tmc2130 stepper_x]
#cs_pin: P1.17
#spi_software_miso_pin: P0.5
#spi_software_mosi_pin: P4.28
#spi_software_sclk_pin: P0.4
##diag1_pin: P1.29
#microsteps: 16
#run_current: 0.800
#hold_current: 0.500
#stealthchop_threshold: 250

#[tmc2130 stepper_y]
#cs_pin: P1.15
#spi_software_miso_pin: P0.5
#spi_software_mosi_pin: P4.28
#spi_software_sclk_pin: P0.4
##diag1_pin: P1.27
#microsteps: 16
#run_current: 0.800
#hold_current: 0.500
#stealthchop_threshold: 250

#[tmc2130 stepper_z]
#cs_pin: P1.10
#spi_software_miso_pin: P0.5
#spi_software_mosi_pin: P4.28
#spi_software_sclk_pin: P0.4
##diag1_pin: P1.25
#microsteps: 16
#run_current: 0.650
#hold_current: 0.450
#stealthchop_threshold: 30

#[tmc2130 extruder]
#cs_pin: P1.8
#spi_software_miso_pin: P0.5
#spi_software_mosi_pin: P4.28
#spi_software_sclk_pin: P0.4
##diag1_pin: P1.28
#microsteps: 16
#run_current: 0.800
#hold_current: 0.500
#stealthchop_threshold: 5

#[tmc2130 extruder1]
#cs_pin: P1.1
#spi_software_miso_pin: P0.5
#spi_software_mosi_pin: P4.28
#spi_software_sclk_pin: P0.4
##diag1_pin: P1.26
#microsteps: 16
#run_current: 0.800
#hold_current: 0.500
#stealthchop_threshold: 5

########################################
# EXP1 / EXP2 (display) pins
########################################

[display]
lcd_type: hd44780
rs_pin: P1.19
e_pin: P1.18
d4_pin: P1.20
d5_pin: P1.21
d6_pin: P1.22
d7_pin: P1.23
encoder_pins: ^P3.26, ^P3.25
click_pin: ^!P0.28
#kill_pin: ^!<RST>

[output_pin beeper]
pin: P1.30

[board_pins]
aliases:
    # EXP1 header
    EXP1_1=P1.30, EXP1_3=P1.18, EXP1_5=P1.20, EXP1_7=P1.22, EXP1_9=<GND>,
    EXP1_2=P0.28, EXP1_4=P1.19, EXP1_6=P1.21, EXP1_8=P1.23, EXP1_10=<5V>,
    # EXP2 header
    EXP2_1=P0.17, EXP2_3=P3.26, EXP2_5=P3.25, EXP2_7=P1.31, EXP2_9=<GND>,
    EXP2_2=P0.15, EXP2_4=P0.16, EXP2_6=P0.18, EXP2_8=<RST>, EXP2_10=<NC>
    # Pins EXP2_1, EXP2_6, EXP2_2 are also MISO, MOSI, SCK of bus "ssp0"

# See the sample-lcd.cfg file for definitions of common LCD displays.

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [heater_bed]
#*# control = pid
#*# pid_kp = 70.559
#*# pid_ki = 2.570
#*# pid_kd = 484.214
#*#
#*# [extruder]
#*# control = pid
#*# pid_kp = 28.250
#*# pid_ki = 1.018
#*# pid_kd = 195.981

cadriel, 2021-02-22:
jobs is greyed out because klipper hasn't communicated that it's ok. check your virtual_sdcard configuration - and make sure it's working correctly.

closed 2021-02-22 by cadriel

Anton33455, 2021-02-22:
I have two identical images with raspberry Pi4  and no matter which one I
connect to this specific printer does it, but works fine on the other one.

cadriel, 2021-02-22:
I'm not sure what else to say.. this is caused by a printer configuration issue. As mentioned, check your configuration - and ensure moonraker is communicating with klipper.

Anton33455, 2021-02-22:
It may be something to do with type of display, i was thinking something
changed with update.  Both of my printers use the same hardware except
display

cadriel, 2021-02-22:
If you aren't getting errors in Fluidd, then my guess is that it's the virtual_sdcard config.

Anton33455, 2021-02-22:
virtual_sdcard config is just two lines that you copy and paste into your configuration.  I have never changed that.  And what I am trying to explain to you is that I get menu item grayed out before any configuration, just by getting clean install.  Something broke down during the update and my old configuration which used to work no longer works. But never mind I am switching back to Octoprint/Klipper  tired of being your guys' guinea pig.

cadriel, 2021-02-23:
Ok, bye. :)